lpodkalicki/blog #1: Using other buttons went wrong!

tom-4, 2016-07-12:
I tried to change the buttons in the code by JUST changing the CASE numbers (because I have another NEC remote control which doesn't have the same buttons which in the code buttons), but, it didn't work!
Any ideas?

lpodkalicki, 2016-07-14:
Hi, thanks for raising this issue.  I guess that you are trying to use incorrect address/command from you remote. It can be quite difficult when it comes to check the code and address without debugging it. This weekend I hope to post an Attiny13 NEC code analyzer that would help you debug it using Software UART.

tom-4, 2016-07-14:
Hi, first thank u very much for replying fast to this issue :)
I'm new to this stuffs but as I understood that this code is for attiny13 @9.6 MHz and for NEC reomte controls, is it right or I missed anything else?
OBS! I used TSOP1738 as an IR receiver!

lpodkalicki, 2016-07-15:
Thats right. The code is for ATtiny13 internal @9.6 MHz and for NEC proto. 

```
$ make
avr-gcc -std=c99 -Wall -g -Os -mmcu=attiny13 -DF_CPU=9600000 -I. -o main.o main.c
avr-ld -o main.elf main.o
avr-objcopy -j .text -j .data -O ihex main.o main.hex	
avr-size -C --mcu=attiny13 main.elf
AVR Memory Usage
----------------
Device: attiny13

Program:     766 bytes (74.8% Full)
(.text + .data + .bootloader)

Data:         15 bytes (23.4% Full)
(.data + .bss + .noinit)

$ make fuse
avrdude -p attiny13 -c usbasp -U hfuse:w:0xFF:m -U lfuse:w:0x7A:m
(...)
$ make flash
avrdude -p attiny13 -c usbasp -U flash:w:main.hex:i -F -P usb
(...)

```

I suggest you to test your project with simpler main function that will catch any signal from your remote (NEC) and indicate a received command/address by toggled led. Example code below.

```
(...)

int
main(void)
{
        uint8_t addr, cmd;

        /* setup */
        DDRB |= _BV(LED_PIN); // set LED pins as OUTPUT
        IR_init();

        /* loop */
        while (1) {
                if (IR_read(&addr, &cmd) == IR_SUCCESS) {
                        PORTB ^= _BV(LED_PIN); // toggle LED1                        
                }
        }
}

```
Hope it will help. L

lpodkalicki, 2016-07-17:
Hi, if you still have a problem with that please take a look at http://blog.podkalicki.com/attiny13-ir-receiver-nec-proto-analyzer/
/L

closed 2016-07-17 by lpodkalicki

tom-4, 2016-07-17:
Hi, thank u very much for all those informations. Actually, I'm using Arduino as an ISP programmer, but, great job thx !

tom-4, 2016-07-30:
Hi again, I tried a lot but it didn't work. I burnt the fuses as you explained. I guess my problem is with the buttons which I defined. My remote control is the Adafruit NEC protocol 44 buttons.
The buttons are like this:
0xFF02FD (0x40)
0xFF1AE5 (0x58)
0xFF3AC5 (0x5C)

........................... and so on, so can you help me please and tell me what I have to change in you code so it works with my remote control, I'm sure that I have to change the cmd cases in the end of the code.
I tried like this, but it didn't work:

while (1) {
		if (IR_read(&addr, &cmd) == IR_SUCCESS) {
			if (addr != 0xff)
				continue;
			switch(cmd) {
			case 0x40:
				PORTB &= ~(_BV(LED1_PIN)|_BV(LED2_PIN)|_BV(LED3_PIN)|_BV(LED4_PIN)); 
				break;
			case 0x58:
				PORTB ^= _BV(LED1_PIN); 
				break;
			case 0x59:
				PORTB ^= _BV(LED2_PIN);
				break;
			case 0x45:
                                PORTB ^= _BV(LED3_PIN); 
                                break;
			case 0x44:
				PORTB ^= _BV(LED4_PIN); 
                                break;
			default:
				break;
			}
		}
	}

lpodkalicki, 2016-08-02:
Hi there. Sorry for late response. Did you try my snippet code from my last post ? I recommend it as a first step of debugging. It will test that your circuit is correct and that library work fine. Expected result is blinking LED when you push _any_ button of your remote. 

`(...)

#define LED_PIN PB0

int
main(void)
{
        uint8_t addr, cmd;

        /* setup */
        DDRB |= _BV(LED_PIN); // set LED pins as OUTPUT
        IR_init();

        /* loop */
        while (1) {
                if (IR_read(&addr, &cmd) == IR_SUCCESS) {
                        PORTB ^= _BV(LED_PIN); // toggle LED                        
                }
        }
}

`