lpodkalicki/blog #5: rainbow_on_several_leds_ws2812

popt2t, 2019-01-12:
When I try to compile the code for Attiny13, I get this message:
`In file included from main.c:8:
c:/winavr-20100110/lib/gcc/../../avr/include/util/delay.h:85:3: warning: #warning "F_CPU not defined for <util/delay.h>"
E:\Temp/ccjZWPjS.o: In function `main':
D:\AVR\blog-master\avr\attiny13\035_rainbow_on_several_leds_ws2812/main.c:42: undefined reference to `ws2812_setleds'`
Please help me.

Ssupreme, 2019-01-12:
Try to put the inline functions into the header file.
And add #define F_CPU 1200000UL .))

popt2t, 2019-01-18:
> Try to put the inline functions into the header file.
How can i do this? I'm a newbie.
> And add #define F_CPU 1200000UL .))
that's ok

lpodkalicki, 2019-02-12:
Thanks @Ssupreme for supporting this case. I'm closing the issue..

closed 2019-02-12 by lpodkalicki