lpodkalicki/blog #12: Problems compiling project. (dht11_temperature_and_humidity_using_software_uart )

Dimitry-Muzi, 2020-10-04:
When trying to compile, it gives two errors.
First:

'int8_t dht_await_state(uint8_t)' was declared 'extern' and later 'static' [-fpermissive]

Second:

undefined reference to `uart_putu(unsigned int)'

The file uart.h contains the line ** void uart_putu (uint16_t x); **, but the void itself is nowhere to be found.

lpodkalicki, 2020-10-05:
Hi, I have no idea what is your environment setup. It compiles, as usual, when you use that one - https://blog.podkalicki.com/how-to-compile-and-burn-the-code-to-avr-chip-on-linuxmacosxwindows/

```
030_log_dht11_temperature_and_humidity_using_software_uart$ make
avr-gcc -std=c99 -Wall -g -Os -mmcu=attiny13 -DF_CPU=1200000 -I. -o main.o main.c uart.c
avr-ld -o main.elf main.o
avr-objcopy -j .text -j .data -O ihex main.o main.hex
avr-size -C --mcu=attiny13 main.elf
AVR Memory Usage
----------------
Device: attiny13

Program:     578 bytes (56.4% Full)
(.text + .data + .bootloader)

Data:         16 bytes (25.0% Full)
(.data + .bss + .noinit)

```

Dimitry-Muzi, 2020-10-05:
Thanks. I changed the compiler - it works. But now it does not show data - everything is zeros. Maybe due to the fact that the sensor is DHT-22?

lpodkalicki, 2020-10-05:
I believe the protocol is the same for both but I didn't check it.

Dimitry-Muzi, 2020-10-05:
I would really appreciate it if you could check.

lpodkalicki, 2020-10-05:
I think you can manage it by comparing this datasheets:
https://www.sparkfun.com/datasheets/Sensors/Temperature/DHT22.pdf
https://components101.com/sites/default/files/component_datasheet/DHT11-Temperature-Sensor.pdf

closed 2020-10-05 by Dimitry-Muzi